PRBonn/semantic_suma #2: How to find the RangeNet_lib ?

Gatsby23, 2019-11-03:
I'm sorry, I have already install the rangenet lib. The package have successfully installed, but when I install the suma++ on the Ubuntu. The "catkin deps fetch" tells me that the rangetnet_lib can't find. Like: 
 [rangenet_lib]           : [NOT FOUND]  

I build the range net in another workspace, but I have already source it and move the build file into the  system lib file. But it still didn't work. Is there some operation like "sudo make install" I should do ? 
I'm sorry to bother you about this. Really thank you for your help.

jbehley, 2019-11-03:
I added the correct url to the package.xml. It should now download everything.

Just run `catkin deps fetch`.

Otherwise, you can also find the github repository under https://github.com/PRBonn/rangenet_lib

Gatsby23, 2019-11-04:
> I added the correct url to the package.xml. It should now download everything.
> 
> Just run `catkin deps fetch`
> 
> Otherwise, you can also find the github repository under https://github.com/PRBonn/rangenet_lib

Do you mean git clone the rangenet_lib under the src folder in the catkin workspace ?

Gatsby23, 2019-11-04:
> I added the correct url to the package.xml. It should now download everything.
> 
> Just run `catkin deps fetch`
> 
> Otherwise, you can also find the github repository under https://github.com/PRBonn/rangenet_lib

Thank you very much, it can work now

jbehley, 2019-11-04:
@Gatsby23, since it is now solved, I simply close this issue.

Thanks for reporting this issue.

closed 2019-11-04 by jbehley