PRBonn/semantic_suma #5: ./visualizer:segmentation fault

xuguangyun, 2019-11-10:
when i run ./visualizer ,there are appear an error: Segmentation fault(core dumped);
I have a default.xml in  ../config folder, and I chang the default.xml : model_path to my path and model files;
So i do not know where the error come from, someone could help me ,thanks!

Chen-Xieyuanli, 2019-11-10:
Thanks for using our code.

Have you already tried the example demo of rangenet_lib?

You may test the rangnet_lib first. It could be the problem of tensorrt of cuda version.

You could find more details here: https://github.com/PRBonn/rangenet_lib

xuguangyun, 2019-11-10:
Thanks for the reply. Yes, the Rangnet_lib have the same error also. 
Here is my environment：Ubuntu18.04, TensorRT 6.01+cuda10.1.

Chen-Xieyuanli, 2019-11-10:
It would be much better if you could provide more details about the compiling and error message/log.

xuguangyun, 2019-11-10:
This is the output of Rangnet_lib :

================================================================================
scan: /root/catkin_ws5/src/rangenet_lib/example/000000.bin
path: /root/catkin_ws5/src/rangenet_lib/example/model/
verbose: 1
================================================================================
Setting verbosity to: false
Trying to open model
Trying to deserialize previously stored: /root/catkin_ws5/src/rangenet_lib/example/model//model.trt
Could not deserialize TensorRT engine. 
Generating from sratch... This may take a while...
Trying to generate trt engine from : /root/catkin_ws5/src/rangenet_lib/example/model//model.onnx
Segmentation fault (core dumped)

xuguangyun, 2019-11-10:
This is the output of when I run the Rangnet_lib:

scan: /root/catkin_ws5/src/rangenet_lib/example/000000.bin
path: /root/catkin_ws5/src/rangenet_lib/example/model/
verbose: 1

Setting verbosity to: false
Trying to open model
Trying to deserialize previously stored: /root/catkin_ws5/src/rangenet_lib/example/model//model.trt
Could not deserialize TensorRT engine. 
Generating from sratch... This may take a while...
Trying to generate trt engine from : /root/catkin_ws5/src/rangenet_lib/example/model//model.onnx
Segmentation fault (core dumped)

xuguangyun, 2019-11-10:
I have debug to the error line: netTensorRT.cpp, line 509 builder->platformHasFastFp16();
the value of "builder" is 0;
The error from TensorRT is almost sure,but how can I  solved this .
Thanks.

Chen-Xieyuanli, 2019-11-11:
Yes, it seems to be a problem caused by the TensorRT.

I didn't meet such a problem so far. I just googled it. It could be a problem with the mismatching of your graphic card and the TensorRT version.

Could you please also report your graphic card and the corresponding driver setups?

You may also give it a try with another version TensorRT, e.g. TensorRT-5.0.

Best

xuguangyun, 2019-11-13:
Thanks. I have try with TensorRT5.1 version,  the last error had disappear,while another error occur >--<:
Here is the terminal output:
--
OpenGL Context version:4.6 core profile
GLEW initialized
OpenGL context version:4.6
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string GeForce GTX 750/PCIe/SSE2
terminate called after throwing an instance of "rv::XmlError"
what(): Error while parsing in line 1
Aborted(core dumped)
--

xuguangyun, 2019-11-13:
Thanks. I have try with TensorRT5.1 version, the last error had disappear,while another error occur.......
Here is the terminal output:
OpenGL Context version:4.6 core profile
GLEW initialized
OpenGL context version:4.6
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string GeForce GTX 750/PCIe/SSE2
terminate called after throwing an instance of "rv::XmlError"
what(): Error while parsing in line 1
Aborted(core dumped)

Chen-Xieyuanli, 2019-11-21:
Sorry for the late reply. It seems it cannot find the configuration file or something wrong when it parses the file.
You could try to run the visualizer with a specific path of the configuration file, like:
./visualizer /path/to/config/file

Chen-Xieyuanli, 2019-12-03:
Hi @xuguangyun

Is there any update about this issue?

xuguangyun, 2019-12-03:
Thanks a lot! I have try to run the visualizer with a specific path of the configuration file, the problem solved！

closed 2019-12-03 by xuguangyun

Dhruv2012, 2020-05-06:
> Sorry for the late reply. It seems it cannot find the configuration file or something wrong when it parses the file.
> You could try to run the visualizer with a specific path of the configuration file, like:
> ./visualizer /path/to/config/file

Hi @Chen-Xieyuanli  I am facing the same issue. I have successfully built the RangenetLib. Could you specify which config file?
It displays this output while running the visualizer script.

**OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce 940MX/PCIe/SSE2
Segmentation fault (core dumped)**

Chen-Xieyuanli, 2020-06-23:
> > Sorry for the late reply. It seems it cannot find the configuration file or something wrong when it parses the file.
> > You could try to run the visualizer with a specific path of the configuration file, like:
> > ./visualizer /path/to/config/file
> 
> Hi @Chen-Xieyuanli I am facing the same issue. I have successfully built the RangenetLib. Could you help out? In this case it doesnot seem to a issue of file location.
> It displays this output while running the visualizer script.
> 
> **OpenGL Context Version 4.6 core profile GLEW initialized. OpenGL context version: 4.6 OpenGL vendor string : NVIDIA Corporation OpenGL renderer string: GeForce 940MX/PCIe/SSE2 Segmentation fault (core dumped)**

Hey @Dhruv2012, sorry for the late reply. I never encountered such a problem, but there are some solutions provide by other users #7. It seems to be a problem caused by GTSAM.