PRBonn/semantic_suma #6: Different semantic results of suma++

TT22TY, 2019-11-20:
Hello,

Thanks for your work！ After I build  the suma++, I just obtain the result like this, I use the pretrained model provided on the webpage, however, it seems that the semantic result is quite different from the picture you provided. I am wondering why it happened( Is it due to the pretrained model used?). I try to use the trained model which is trained from scratch by myself, it works well using the infer.py of RangeNet++, the result is also strange. Besides, the result of RangeLib are also different from the picture you provided. 
Looking forward from your reply. :) Thanks!

[image]

Chen-Xieyuanli, 2019-11-21:
Hey,

Thanks for using our code. 

It looks like a problem of the semantic segmentation part, and the SLAM part works. You could visualize the semantic segmentation results by checking the box of "Show semantic map"

Several reasons may cause this problem.
1. Our rangenet_lib and the provided pretrained model can now only work well with KITTI LiDAR data. It may provide wrong predictions when you test with other LiDAR data.
2. The runtime tensorrt model will be generated depending on your hardware. It may not work with other versions of the graphic cards or drivers. Could you please provide more information about your hardware setups. We may also have a test later.

TT22TY, 2019-11-25:
Hi, 
Thanks for your reply!
Yes, it seems that it is a problem of the semantic segmentation part. As I check the box of "Show semantic map", it looks like this, in which the semantic map looks black with only a bit of color labels.
[image]

I check the project on my machine with regard to your suggestions:
[image]

1. I do use the KITTI LiDAR data for testing, specifically, I use both the KITTI  odometry velodyne laser data (The semantic kitti data set your lab provide) and road data (2011_10_03_drive_0047_sync) as you mentioned in your SuMa++ paper, both of them did not work correctly.

2. My hardware set up  is as follows:
[image]
I use GeForce RTX 2080,Cuda 10.0, V10.0.130, Cudnn 7.5.0,  Driver Version: 410.48. 

3. Besides, when I build the project "point labeler"(https://github.com/PRBonn/point_labeler) and "SuMa ++" , gtest related problems occurred, 
 after I remove gtest with "apt-get purge libgtest-dev " , the point labeler works fine.

 However , I am not sure it will affect the semantic map, since when I build suma++. it will give the prompt message: "gtest not found, C++ tests can not be built. Please install the gtest headers globally in your system to enable gtest"
 
[image]

Looking forward to your reply. :) Thanks!

Chen-Xieyuanli, 2019-11-25:
The problem should come from the rangenet_lib, since the semantic segmentation input of SuMa++ is not correct.

Have you already tried the example demo of rangenet_lib? The semantic segmentation result visualized in the "Show semantic map" should be the same as that of rangenet_lib.

You may test and make sure the rangnet_lib works first. If the example demo of rangenet_lib works well, SuMa++ should also work, and vice versa.

TT22TY, 2019-11-25:
Thanks for your reply!

Yes, I have tested rangenet_lib, the result is as follows, which is also different to the demo on the website(https://github.com/PRBonn/rangenet_lib). I use the pre_trained model provided on the website, I wonder why it is different and wrong. Could you please give some suggestions to find out the reason?
[image]

 Thank you very much.

Chen-Xieyuanli, 2019-12-03:
Hi @TT22TY,
Since the problem comes from rangenet_lib and you also opened a new issue there, I'm going to close this one.

Chen-Xieyuanli, 2019-12-05:
Hi @TT22TY 

The problem seems caused by the incompatibility of our code with TensorRT version 6. 

For more details please find here: https://github.com/PRBonn/rangenet_lib/issues/9.

TT22TY, 2019-12-05:
Thank you very much! @Chen-Xieyuanli  

In fact ,I use TensorRT 5.1.5.0, so I wonder which version do you installed?
Besides, since the version of TensorRT is related to cuda and cudnn, could you please tell me the exact version of the cuda and cudnn you use? Mine is Cuda 10.0, V10.0.130, Cudnn 7.5.0, Driver Version: 410.48. TensorRT 5.1.5.0. 

Thank you!

Chen-Xieyuanli, 2019-12-05:
Hi @TT22TY ,

I've listed the tested setups in https://github.com/PRBonn/rangenet_lib/issues/9.

To keep this issue in one track, I am going to close this one again.

closed 2019-12-05 by Chen-Xieyuanli

G12311231, 2024-07-02:
> Hey,
>
> Thanks for using our code.
>
> It looks like a problem of the semantic segmentation part, and the SLAM part works. You could visualize the semantic segmentation results by checking the box of [...]

Hi, 
If I use the same sensor vlp-64 to collect data on my campus, I wonder if this dataset will be able to use the parameters of the pre-trained model?