PRBonn/semantic_suma #7: The visualizer crashes

MohamedAfifii, 2019-11-24:
The visualizer crashes giving different output each time I run it (without making any changes).

Segmentation fault
```
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080 Ti/PCIe/SSE2
Segmentation fault (core dumped)
```

rv::XmlError
```
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080 Ti/PCIe/SSE2
terminate called after throwing an instance of 'rv::XmlError'
  what():  Error while parsing in line 1
Aborted (core dumped)
```

Sometimes I also reach 
```
Extracting surfel maps partially.
Performing frame-to-model matching.
```
But it also crashes.

---

- `rangenet_lib` works fine with me. 

- As for the build I used:
```
catkin build --save-config -i --cmake-args -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES
```

- And here is  the output of `glxinfo | grep "version"`
```
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.5.0 NVIDIA 390.25
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 390.25
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 390.25
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix, 
```

- I just had to change a line in the `include_directories` in the `CMakeLists.txt` from `/usr/include/eigen3` to `/usr/local/include/eigen3` because I got the following error:
```
error: static assertion failed: Error: GTSAM was built against a different version of Eigen
```
which made the build work.

Chen-Xieyuanli, 2019-11-25:
Thanks for using our code. This issue seems similar to #5. You may have a look.

MohamedAfifii, 2019-11-25:
Thanks for your reply.

I tried `./bin/visualizer ./config/default.xml`. The `rv::XmlError` doesn't appear but I still get segmentation fault.

Also now sometimes the GUI opens for a small amount of time (less than a second) then it crashes.

As for the `default.xml`, I changed only the last line to
```
<param name="model_path" type="string">/home/ubi-comp/Downloads/darknet53</param>
```

Also the build of `semantic_suma` finishes with 3 warnings. I can post them here if they may add additional information.

MohamedAfifii, 2019-11-25:
There is another point that might be relevant. 

Inside `tensorrt-config.cmake` of `rangenet_lib` the build was not able to `find_library` the components of TensorRT. So I had to manually add these two lines:
```
include_directories(/home/ubi-comp/Downloads/TensorRT-5.1.5.0/include)
link_directories(/home/ubi-comp/Downloads/TensorRT-5.1.5.0/lib)
```

And to comment out `set(TENSORRT_FOUND OFF)`. This made `rangenet_lib` work properly.

Chen-Xieyuanli, 2019-12-03:
Hi @MohamedAfifii,

Is there any update about this issue?

MohamedAfifii, 2019-12-09:
The problem was mainly with the GTSAM version. I had to use version `4.0.0-alpha2`.

closed 2019-12-09 by MohamedAfifii

Claud1234, 2020-06-22:
> The problem was mainly with the GTSAM version. I had to use version `4.0.0-alpha2`.

Hi. I think I have the same problem about 'segmentation fault'. I have confirmed that my rangenet_lib works good and its result is identical to template. 

Sometimes I also can reach the lines like this:
[image]
and sometimes there is some graphic window showed up but crashed quickly.

So my question is that is the version of GTSAM really important over here? I have the latest 4.0.2 installed in my system. Or am I missing something else? @Chen-Xieyuanli

Chen-Xieyuanli, 2020-06-23:
Hey @Claud1234, I never encountered such a problem, but it seems there are also some reasons which may cause the crash. You may have a look at #5 and [here](https://github.com/jbehley/SuMa/issues/17)

Claud1234, 2020-06-23:
> Hey @Claud1234, I never encountered such a problem, but it seems there are also some reasons which may cause the crash. You may have a look at #5 and [here](https://github.com/jbehley/SuMa/issues/17)

Okay. Thanks for reply first. I will solve my own problems here. The version of GTSAM exactly is the reason that result visualizer crashed. I had the latest one 4.0.2 before, everything works good after I downgrade it to 4.0.0-alpha-2.

Thanks for help anyway!