PRBonn/semantic_suma #9: Invalid framebuffer object. Code 36061(GL_FRAMEBUFFER_UNSUPPORTED)

zzhh00, 2019-12-10:
Hi,
I have a nvidia graphic card which is GeForce GT730,as be shown below:
[image]
The visualizer also crashes for the error.Can you tell me how can I solve this issue please?Thank you very much!

zzhh00, 2019-12-10:
[image]
The problem may be caused by this.I added --no-opengl-files when installing nvidia driver.However,someone said that the computer can not enter the graphic mode if installing nvidia drvier withou --no-opengl-files.What should I do then?

Chen-Xieyuanli, 2019-12-10:
Hi @zzhh00,

Thanks for using our code. 

Several reasons could cause the visualizer crashes. You may have a look at similar issues #5 and #7 first.

It would be also helpful if you could provide the information about executing $ glxinfo | grep OpenGL

zzhh00, 2019-12-11:
Hi@Chen-Xieyuanli,
Thanks for your reply.

The information about OpenGL is list as below and I am trying to change the OpenGL version.
[image]

zzhh00, 2019-12-11:
Besides when type nvidia-settings it shows that ERROR: Unable to load info from any available system.Thus maybe the nvidia driver installation is failed.

zzhh00, 2019-12-11:
1.I solved this problem as the solution referred in Suma issue #2.However this solution is aim at the one  who only has Intel graphics and I have nvidia graphic card even though there may be something wrong.
2.Suma++ miss .obj files for visualization and these files can be found in assert of Suma.

jbehley, 2019-12-11:
@zzhh00 

1. Your system currently uses the Intel GPU (from your CPU) and if you want to use your nvidia graphics card, you definitly have to change something. The OpenGL vendor string should contain something with NVIDIA, something like:

```
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 960/PCIe/SSE2
```
In theory, the code should also work with an Intel GPU, but since NIVIDA usually provides more in their driver than the OpenGL Core Profile asks for it is hard for us to test that everything would run with the minimum requirements (since we use an NVIDIA driver for testing). 

2. The second problem was caused by missing files in the repository. A `git pull`should give you the needed files in the folder `assets`.

Chen-Xieyuanli, 2019-12-26:
Since the problem has been solved, I am going to close this issue. If there are any updates, please feel free to ask me to reopen it.

closed 2019-12-26 by Chen-Xieyuanli

lifangyan16, 2022-01-21:
> @zzhh00
>
>     1. Your system currently uses the Intel GPU (from your CPU) and if you want to use your nvidia graphics card, you definitly have to change something. The OpenGL vendor string should [...]

I have this problem too, but I don't know how to change the OpenGL vendor string. I have installed the NVIDIA-Linux-x86_64-470.94.run, and the information is on below.
/home/lfy/图片/2022-01-21 19-56-20屏幕截图.png