PRBonn/semantic_suma #10: Failed to run visualizer...

LongruiDong, 2019-12-19:
HI~

after building the project, and run ./visualizer , I got this error 

```
OpenGL Context Version 3.3 core profile
GLEW initialized.
OpenGL context version: 3.3
OpenGL vendor string  : VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 6.0, 256 bits)
terminate called after throwing an instance of 'glow::GlShaderError'
  what():  shader/gen_surfels.geom: 0:1(10): error: GLSL 4.00 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES

Aborted (core dumped)

```
Here is output when run  glxinfo | grep "version", I have set  -DOPENGL_VERSION=330 to compile...
```
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 3.3
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.0.5
OpenGL core profile shading language version string: 3.30
OpenGL version string: 3.0 Mesa 18.0.5
OpenGL shading language version string: 1.30
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.0.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00

```
do you know how to pass it?

Thanks ~

jbehley, 2019-12-19:
Sorry for the trouble. We encountered this issue now multiple times.

It seems that we use features that are only available with version 4.0 of OpenGL and therefore I adjustet the minimum requirement.

We currently have also to figure out if this problem persists with version OpenGL 4.0 (Core Profile), but we never encountered issues with NVIDIA GPUs so far, since their driver supports actually more then the Core Profile requires.

One issue might be also running the code in a virtual machine. Currently, we never experienced so much problems with ubunutu.

LongruiDong, 2019-12-19:
> Sorry for the trouble. We encountered this issue now multiple times.
>
> It seems that we use features that are only available with version 4.0 of OpenGL and therefore I adjustet the minimum [...]

OK, I am going to update my OpenGL to >=4.0 to have a try

Chen-Xieyuanli, 2019-12-26:
We mostly tested our code with Nvidia GPUs and it works well with OpenGL 4.0.

When using with other graphic cards, problems may happen because Nvidia drivers support more than OpenGL core profile.

We currently don't have any plans to test with other graphic cards. We will appreciate it a lot if you want to test and report it here.

Therefore, I will close this issue and please feel free to ask me to reopen it.

Thanks in advance.

closed 2019-12-26 by Chen-Xieyuanli