PRBonn/semantic_suma #12: CMake Error when compiling...

LongruiDong, 2019-12-25:
Hi, 
When I compile the project, i received the following errors:

```
......
[semantic_suma:cmake] -- Using OpenGL version 450.                                                                                                
[semantic_suma:cmake] -- Configuring done                                                                                                         
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:175 (add_library):                                 
[semantic_suma:cmake]   Target "suma" links to target "Boost::serialization" but the target was not                                               
[semantic_suma:cmake]   found.  Perhaps a find_package() call is missing for an IMPORTED target, or                                               
[semantic_suma:cmake]   an ALIAS target is missing?                                                                                               
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:175 (add_library):                                 
[semantic_suma:cmake]   Target "suma" links to target "Boost::thread" but the target was not found.                                               
[semantic_suma:cmake]   Perhaps a find_package() call is missing for an IMPORTED target, or an                                                    
[semantic_suma:cmake]   ALIAS target is missing?                                                                                                  
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:175 (add_library):                                 
[semantic_suma:cmake]   Target "suma" links to target "Boost::date_time" but the target was not                                                   
[semantic_suma:cmake]   found.  Perhaps a find_package() call is missing for an IMPORTED target, or                                               
[semantic_suma:cmake]   an ALIAS target is missing?                                                                                               
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:175 (add_library):                                 
[semantic_suma:cmake]   Target "suma" links to target "Boost::regex" but the target was not found.                                                
[semantic_suma:cmake]   Perhaps a find_package() call is missing for an IMPORTED target, or an                                                    
[semantic_suma:cmake]   ALIAS target is missing?                                                                                                  
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:175 (add_library):                                 
[semantic_suma:cmake]   Target "suma" links to target "Boost::timer" but the target was not found.                                                
[semantic_suma:cmake]   Perhaps a find_package() call is missing for an IMPORTED target, or an                                                    
[semantic_suma:cmake]   ALIAS target is missing?                                                                                                  
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:175 (add_library):                                 
[semantic_suma:cmake]   Target "suma" links to target "Boost::chrono" but the target was not found.                                               
[semantic_suma:cmake]   Perhaps a find_package() call is missing for an IMPORTED target, or an                                                    
[semantic_suma:cmake]   ALIAS target is missing?                                                                                                  
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:156 (add_library):                                 
[semantic_suma:cmake]   Target "robovision" links to target "Boost::serialization" but the target                                                 
[semantic_suma:cmake]   was not found.  Perhaps a find_package() call is missing for an IMPORTED                                                  
[semantic_suma:cmake]   target, or an ALIAS target is missing?                                                                                    
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:156 (add_library):                                 
[semantic_suma:cmake]   Target "robovision" links to target "Boost::thread" but the target was not                                                
[semantic_suma:cmake]   found.  Perhaps a find_package() call is missing for an IMPORTED target, or                                               
[semantic_suma:cmake]   an ALIAS target is missing?                                                                                               
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:156 (add_library):                                 
[semantic_suma:cmake]   Target "robovision" links to target "Boost::date_time" but the target was                                                 
[semantic_suma:cmake]   not found.  Perhaps a find_package() call is missing for an IMPORTED                                                      
[semantic_suma:cmake]   target, or an ALIAS target is missing?                                                                                    
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake]                                                                                                                             
[semantic_suma:cmake] CMake Error at /home/dlr/Project/suma++/src/semantic_suma/CMakeLists.txt:156 (add_library):                                 
[... cut ...]
ii  python3-libnvinfer                                          5.1.5-1+cuda10.0                                      amd64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                                      5.1.5-1+cuda10.0                                      amd64        Python 3 development package for TensorRT
ii  tensorrt                                                    5.1.5.0-1+cuda10.0                                    amd64        Meta package of TensorRT

```

How can I solve it?

Thanks a lot~

Chen-Xieyuanli, 2019-12-26:
Hey @LongruiDong,

Thank you for using our code. As can be read from the Cmake Error message, it seems to be a problem of the Boost library. You could check your Boost library by executing $ dpkg -s libboost-dev | grep 'Version'

Chen-Xieyuanli, 2020-01-09:
Hi @LongruiDong, is there any update on this issue?

Chen-Xieyuanli, 2020-01-20:
Since there is no update for this issue, I will then close it.

If needed, feel free to ask me to reopen this issue.

closed 2020-01-20 by Chen-Xieyuanli

KangchengLiu, 2020-08-27:
How was the probem solved?

he-guo, 2021-05-18:
Hi, I have the same problem, and use:
$ dpkg -s libboost-dev | grep 'Version':Version: 1.65.1.0ubuntu1
what else can I do? thanks.  @Chen-Xieyuanli

he-guo, 2021-05-19:
> Hi, I have the same problem, and use:
> $ dpkg -s libboost-dev | grep 'Version':Version: 1.65.1.0ubuntu1
> what else can I do? thanks. @Chen-Xieyuanli

I have solved this problem, and I change line 9 in CMakeLists.txt:
    find_package(Boost 1.65.1 REQUIRED COMPONENTS filesystem system serialization thread date_time regex timer chrono)

nkymzsy, 2021-07-11:
> I have solved this problem, and I change line 9 in CMakeLists.txt:
> find_package(Boost 1.65.1 REQUIRED COMPONENTS filesystem system serialization thread date_time regex timer chrono)

It also slove my problem!

FALCONYU, 2023-04-27:
It also slove my problem!