PRBonn/semantic_suma #13: build error

TT22TY, 2020-01-21:
Hi, @Chen-Xieyuanli 
Thanks for your code~
I tried on my computer two month ago, it can work normally. Now  I just try to run it on Xavier, but some errors occur,  I just wonder is it related to the Xavier I used, since it is different from the common computer. Can this project be used on Xavier?

[image]

Thanks very much.

Chen-Xieyuanli, 2020-01-29:
Hey @TT22TY, 

Sorry for the late reply. We currently have no plan to try it on Xavier. From the error msg, there seems a problem in your GLEW library. You may have a look at GLEW (https://github.com/nigels-com/glew)

TT22TY, 2020-02-04:
@Chen-Xieyuanli  Thank you very much!

closed 2020-02-04 by TT22TY