PRBonn/semantic_suma #14: visualizer segmentation fault

undefinedzero, 2020-03-13:
I used the command `./visualizer ~/catkin_suma/src/semantic_suma-master/config/default.xml` to run the visualizer, but I encountered a segmentation fault.
```
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
[1]    21658 segmentation fault (core dumped)  ./visualizer
```
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc42d1700 (LWP 21782)]
[New Thread 0x7fffb9d9e700 (LWP 21783)]
[New Thread 0x7fffb959d700 (LWP 21784)]
[New Thread 0x7fffb379a700 (LWP 21785)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080/PCIe/SSE2

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
0x00005555555b2e39 in _mm256_store_ps (__A=..., __P=<optimized out>)
    at /usr/lib/gcc/x86_64-linux-gnu/7/include/avxintrin.h:879
879	  *(__m256 *)__P = __A;
(gdb) 
```

I tried Rangenet, it seems good.
```
================================================================================
scan: ~/catkin_suma/src/rangenet_lib-master/example/000000.bin
path: ~/catkin_suma/src/semantic_suma-master/darknet53/
verbose: 0
================================================================================
Setting verbosity to: false
Trying to open model
Trying to deserialize previously stored: ~/catkin_suma/src/semantic_suma-master/darknet53//model.trt
Could not deserialize TensorRT engine. 
Generating from sratch... This may take a while...
Trying to generate trt engine from : ~/catkin_suma/src/semantic_suma-master/darknet53//model.onnx
Platform DOESN'T HAVE fp16 support.
No DLA selected.
----------------------------------------------------------------
Input filename:   ~/catkin_suma/src/semantic_suma-master/darknet53//model.onnx
ONNX IR version:  0.0.4
Opset version:    9
Producer name:    pytorch
Producer version: 1.1
Domain:           
Model version:    0
Doc string:       
----------------------------------------------------------------
WARNING: ONNX model has a newer ir_version (0.0.4) than this parser was built against (0.0.3).
 ----- Parsing of ONNX model ~/catkin_suma/src/semantic_suma-master/darknet53//model.onnx is Done ---- 
Success picking up ONNX model
Failure creating engine from ONNX model
Current trial size is 8589934592
Cuda error in file src/implicit_gemm.cu at line 648: out of memory
Success creating engine from ONNX model
Final size is 4294967296
Success creating engine from ONNX model
Trying to serialize engine and save to : ~/catkin_suma/src/semantic_suma-master/darknet53//model.trt for next run
Binding: 0, type: 0
[Dim 5][Dim 64][Dim 2048]
Binding: 1, type: 0
[Dim 20][Dim 64][Dim 2048]
Successfully create binding buffer
================================================================================
Predicting image: ~/catkin_suma/src/rangenet_lib-master/example/000000.bin
================================================================================
Example finished! 
```

undefinedzero, 2020-03-13:
Here are some outputs from valgrind
```
$ valgrind ./visualizer
==8880== Memcheck, a memory error detector
==8880== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==8880== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==8880== Command: ./visualizer
==8880== 
==8880== Warning: set address range perms: large range [0x16ee1000, 0x2c4fa000) (defined)
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080/PCIe/SSE2
here
==8880== Warning: set address range perms: large range [0x59e43040, 0x69e43087) (undefined)
==8880== Warning: set address range perms: large range [0x69e44040, 0x79e44087) (undefined)
Extracting surfel maps partially.
==8880== Warning: set address range perms: large range [0x79e45040, 0x8b0ed840) (undefined)
Performing frame-to-model matching.
==8880== Warning: set address range perms: large range [0x8b0ee040, 0xd2956c87) (undefined)
==8880== Conditional jump or move depends on uninitialised value(s)
==8880==    at 0x530F335: glow::RoSeCamera::matrix() (in /home/bill/catkin_suma/devel/.private/glow/lib/libglow_util.so)
==8880==    by 0x167D39: ViewportWidget::paintGL() (ViewportWidget.cpp:561)
==8880==    by 0x637DCC3: QGLWidget::glDraw() (in /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5.9.5)
==8880==    by 0x637D66C: QGLWidget::paintEvent(QPaintEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5.9.5)
==8880==    by 0x6747047: QWidget::event(QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x670883B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6710103: QApplication::notify(QObject*, QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x7E0A8D7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5)
==8880==    by 0x6740199: QWidgetPrivate::sendPaintEvent(QRegion const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6740759: QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6715F5A: QWidgetPrivate::repaint_sys(QRegion const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6763D4D: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880== 
==8880== 
==8880== Process terminating with default action of signal 11 (SIGSEGV)
==8880==  General Protection Fault
==8880==    at 0x167D3A: _mm256_load_ps (avxintrin.h:873)
==8880==    by 0x167D3A: pload<__vector(8) float> (PacketMath.h:213)
==8880==    by 0x167D3A: ploadt<__vector(8) float, 32> (GenericPacketMath.h:463)
==8880==    by 0x167D3A: packet<32, __vector(8) float> (CoreEvaluators.h:197)
==8880==    by 0x167D3A: assignPacket<32, 32, __vector(8) float> (AssignEvaluator.h:652)
==8880==    by 0x167D3A: assignPacketByOuterInner<32, 32, __vector(8) float> (AssignEvaluator.h:666)
==8880==    by 0x167D3A: run (AssignEvaluator.h:274)
==8880==    by 0x167D3A: run (AssignEvaluator.h:434)
==8880==    by 0x167D3A: call_dense_assignment_loop<Eigen::Matrix<float, 4, 4>, Eigen::Matrix<float, 4, 4>, Eigen::internal::assign_op<float, float> > (AssignEvaluator.h:741)
==8880==    by 0x167D3A: run (AssignEvaluator.h:879)
==8880==    by 0x167D3A: call_assignment_no_alias<Eigen::Matrix<float, 4, 4>, Eigen::Matrix<float, 4, 4>, Eigen::internal::assign_op<float, float> > (AssignEvaluator.h:836)
==8880==    by 0x167D3A: call_assignment<Eigen::Matrix<float, 4, 4>, Eigen::Matrix<float, 4, 4>, Eigen::internal::assign_op<float, float> > (AssignEvaluator.h:804)
==8880==    by 0x167D3A: call_assignment<Eigen::Matrix<float, 4, 4>, Eigen::Matrix<float, 4, 4> > (AssignEvaluator.h:782)
==8880==    by 0x167D3A: _set<Eigen::Matrix<float, 4, 4> > (PlainObjectBase.h:710)
==8880==    by 0x167D3A: operator= (Matrix.h:208)
==8880==    by 0x167D3A: ViewportWidget::paintGL() (ViewportWidget.cpp:561)
==8880==    by 0x637DCC3: QGLWidget::glDraw() (in /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5.9.5)
==8880==    by 0x637D66C: QGLWidget::paintEvent(QPaintEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5.9.5)
==8880==    by 0x6747047: QWidget::event(QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x670883B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6710103: QApplication::notify(QObject*, QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x7E0A8D7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.9.5)
==8880==    by 0x6740199: QWidgetPrivate::sendPaintEvent(QRegion const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6740759: QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6715F5A: QWidgetPrivate::repaint_sys(QRegion const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x6763D4D: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880==    by 0x676466C: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.9.5)
==8880== 
==8880== HEAP SUMMARY:
[... cut ...]
==8880==                         newarray           : 2,112 bytes in 52 blocks
==8880==                         multipleinheritance: 808 bytes in 5 blocks
==8880==         suppressed: 0 bytes in 0 blocks
==8880== Rerun with --leak-check=full to see details of leaked memory
==8880== 
==8880== For counts of detected and suppressed errors, rerun with: -v
==8880== Use --track-origins=yes to see where uninitialised values come from
==8880== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
[1]    8880 segmentation fault (core dumped)  valgrind ./visualizer
```

jbehley, 2020-03-13:
Problems with Eigen usually originate in clash between gtsam's packaged and the systems Eigen version.

Please try to reinstall gtsam with the compile option `USE_SYSTEM_EIGEN` set.

undefinedzero, 2020-03-13:
I tried recompiling gtsam with systems Eigen(version 3.3.7), but the problem remains. I fixed this problem by changing the Eigen's version to 3.2.10

closed 2020-03-13 by undefinedzero

Chen-Xieyuanli, 2020-03-13:
@undefinedzero Thank you for the feedback!

undefinedzero, 2020-03-13:
I found that using Eigen 3.2.10 will also encounter a segmentation fault when reading calibration files. I tried 3.2.1, it seems good for now.