PRBonn/semantic_suma #15: Failed to initialize GLEW on AGX

NagarajDesai1, 2020-03-19:
I am trying to run suma++ on NVIDIA AGX development toolkit. I am getting the following error when I try to run the visualizer:

QEGLPlatformContext: Failed to create context: 3009
OpenGL Context Version 2.0 compatibility profile
Missing GL version
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to initialize GLEW.
Aborted (core dumped)

GLEW is not initialized, could you provide a solution? I am new to using OpenGL.

jbehley, 2020-03-19:
We rely on features from OpenGL Core Profile 4.0 and above. We currently do no provide support for OpenGL Compatibility Profile or OpenGL ES.

NagarajDesai1, 2020-03-19:
this is the terminal output when i query OpenGL version.
glxinfo | grep "version"
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.6.0 NVIDIA 32.3.1
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 32.3.1
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 32.3.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_group_vote, GL_EXT_shader_implicit_conversions,

jbehley, 2020-03-20:
The problem might be related to: 

https://stackoverflow.com/questions/47415198/missing-gl-version-from-glewinit-using-egl/54668271#54668271

In the second answer the final solution was: Try to "Link against libOpenGL.so instead of libGL.so" by modifying the `CMakefile.txt`. (you have to replace `${OpenGL_LIBRARIES}`with just `OpenGL`.  I don't know if this works, since I could not test it right now.

Unfortunately, we cannot really provide support for AGX.

NagarajDesai1, 2020-03-26:
I am now facing linking error.

`[semantic_suma:make] [100%] Linking CXX executable
[semantic_suma:make] /home/anavs/catkin_ws/src/semantic_suma/bin/visualizer 
/catkin_ws/devel/.private/semantic_suma/lib/libsuma.so: undefined reference to gluErrorString'                                                                             
[semantic_suma:make] collect2: error: ld returned 1 exit status                                                                                                                                             
[semantic_suma:make] CMakeFiles/visualizer.dir/build.make:417: recipe for target '/home/anavs/catkin_ws/src/semantic_suma/bin/visualizer' failed                                                            
[semantic_suma:make] make[2]: *** [/home/anavs/catkin_ws/src/semantic_suma/bin/visualizer] Error 1                                                                                                          
[semantic_suma:make] CMakeFiles/Makefile2:329: recipe for target 'CMakeFiles/visualizer.dir/all' failed                                                                                                     
[semantic_suma:make] make[1]: *** [CMakeFiles/visualizer.dir/all] Error 2                                                                                                                                   
[semantic_suma:make] Makefile:157: recipe for target 'all' failed                                                                                                                                           
[semantic_suma:make] make: *** [all] Error 2                                                         `

What is your hardware platform on which you could run this visualization code?

jbehley, 2020-03-26:
Just normal Ubuntu 18.04 and Ubuntu 16.04. Normal PCs with Nvidia GPUs (1080s, 960, 2070, Quadro P6000, ...).

Chen-Xieyuanli, 2020-04-01:
Hey @NagarajDesai1, I'm sorry that current our code only works on certain hardware platforms since we mainly focus on semantic SLAM approach research and cannot provide much help in the hardware developing side.

I'll therefore close this issue. Please feel free to ask me to reopen it if needed.

closed 2020-04-01 by Chen-Xieyuanli