PRBonn/semantic_suma #16: The error : 'std::runtime_error'

ONMEKEEP, 2020-03-23:
Hello.When I ran the ./visualize and selected the data file, the runing window disappeared. But I can run the rangenet_lib correctly. The following are the tips from command window.
/********************************************/
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce RTX 2060/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Can't open cfg.yaml from ~/test/darknet53/arch_cfg.yaml
/**********************************************/
Hope someone can help me solve this problem! Thanks.

Chen-Xieyuanli, 2020-03-23:
Hey @ONMEKEEP,

As the error msg says it cannot find the yaml file. Could you please check the path to the model in the config file?

ONMEKEEP, 2020-03-23:
> Hey @ONMEKEEP,
> 
> As the error msg says it cannot find the yaml file. Could you please check the path to the model in the config file?

Thank you for your reply. The model is located in /home/test/darknet53. And I modified the last line of default.xml with ~/test/darknet53. But it got the above error.

Chen-Xieyuanli, 2020-03-23:
You could try with the full name rather than '~'. The tilde sometimes will not be parsed correctly.

Chen-Xieyuanli, 2020-03-23:
Thank you for using our code. If you like it, you could star, watch and fork our code.

Since the problem is solved, I'm going to close this issue.

closed 2020-03-23 by Chen-Xieyuanli