PRBonn/semantic_suma #17: Code segment for ICP improvement?

anthonypan08, 2020-03-29:
I would like to build on top of the ICP algorithm by utilizing EM algorithm to soft assign the points. I was curious where exactly are the ICP algorithms implemented. 

I've dugged through the code and it seemed like the ICP algorithm is implemented in src/shaders/Frame2Model_jacobians.geom. However, it seems like it is a OpenGL format so I was wondering if I should modify the algorithm there directly or if I should modify the algorithm elsewhere.

Chen-Xieyuanli, 2020-03-29:
Hey @anthonypan08,

Thanks for using our code. We actually use OpenGL to accelerate the calculation of the jacobians in Frame2Model_jacobians.geom. For the entire frame-to-model ICP process, you may also have a look at Frame2Model.cpp, LieGaussNewton.cpp and also the updatePose() function in SurfelMapping.cpp.

Hope it helps!

anthonypan08, 2020-03-30:
It seem like Frame2Model.cpp utilizes the Frame2Model.geom file and the weights of each point are kept track in Frame2Model.geom . From what I can see, the .xyz symbolizes the 3D coordinate of each point in point cloud and the .w is the weights. Also can you tell me where those properties are defined (I am quite new to OpenGL so forgive me if the questions are elementary)

My plan is to use expectation maximization algorithm to soft assign the weight of each point. Is it correct that I would have to modify the contents in the OpenGL Frame2Model.geom? I want to ensure that each point holds an array of possible weights instead of just 1 weight. Do know any specific issue that would arise if I just go ahead and change the data structure in OpenGL?

Chen-Xieyuanli, 2020-04-01:
Yes, you are right. You could find the definition of Surfel properties in Surfel.h.

Maintaining an array for each point is not that straightforward, but it is definitely doable. You could find more information [here](https://www.khronos.org/opengl/wiki/Data_Type_(GLSL)#Arrays)

Chen-Xieyuanli, 2020-04-28:
Since there is no update, I'm going to close this issue. Please feel free to ask me to reopen it if needed.

closed 2020-04-28 by Chen-Xieyuanli