PRBonn/semantic_suma #19: Debug OpenGL

jerlomy4ever, 2020-04-01:
Is there any method to debug the code for OpenGL, like printing in OpenGL ?

jbehley, 2020-04-01:
There is, unfortunately, no such thing. For debugging you have to write values to textures.

Please refer to the excellent tutorials at https://learnopengl.com/ to learn the basics needed.

Chen-Xieyuanli, 2020-04-01:
Thank @jbehley for the reply. As said, we actually didn't use any OpenGL debugger.

However, there are some OpenGL debuggers ([here](https://www.khronos.org/opengl/wiki/Debugging_Tools)), but I'm not sure whether they are easy to use or not.

You may give a try, and I'll appreciate it a lot if you could provide any feedback about that.

Since we cannot provide more help, I'll close this issue. Please feel free to ask me to reopen it if needed.

closed 2020-04-01 by Chen-Xieyuanli