PRBonn/semantic_suma #20: Not the same as video demo's

topcomma, 2020-04-15:
Hi guys ,

Thanks for your contributions to us. 
Recently I have configured the environment of the SuMa++ working and I have got the results from the Kitti's dataset as your paper's.
[image]
[image]

Here I attach my results(two images), I wonder why not the same result?

Thanks.

Chen-Xieyuanli, 2020-04-15:
Hey @topcomma,

Thanks for using our code. It seems the semantic inferring results are not correct. Before running SuMa++, you may first have a test with the given demo in [rangnet_lib](https://github.com/PRBonn/rangenet_lib) to make sure that the semantic inferring works well.

topcomma, 2020-04-15:
@Chen-Xieyuanli ,thanks for you quick response. 
I will double-check it.
Thanks.

topcomma, 2020-04-15:
@Chen-Xieyuanli ,I ran it again and got the result of executing 'infer' bin, here I provide  the captured screen for '/example/000000.bin' and its log.
[image]
[image]

It seems to work fine.

Chen-Xieyuanli, 2020-04-15:
The inferring result is not correct. As a comparison, you could find the correct one [here](https://github.com/PRBonn/rangenet_lib/blob/master/pics/demo.png).

This problem may be caused by an incompatible TensorRT version. You may try with TensorRT 5.1.0, which works well in our cases.

topcomma, 2020-04-15:
@Chen-Xieyuanli ,thanks.
I will try it again.

topcomma, 2020-04-16:
@Chen-Xieyuanli ,

Today, I have re-installed TensorRT,changed its version(6.1) to version(5.1), but 'infer' still not correct!

[image]

Chen-Xieyuanli, 2020-04-16:
I think this is the same issue as discussed in rangnetlib [issue 9](https://github.com/PRBonn/rangenet_lib/issues/9). Could you please have a look and have a try with the solutions provided there?

To better track the same issue, I would like to close this duplicated issue and stick to the previous one in rangnetlib. 

If there are some other different problems, please feel free to ask me to reopen this issue.

closed 2020-04-16 by Chen-Xieyuanli

topcomma, 2020-04-16:
@Chen-Xieyuanli ,thanks.