PRBonn/semantic_suma #21: Geo-referenced Map

Soini3913, 2020-04-26:
Hi :) Thanks for the nice work. 

As far as i can see you are only using the Pointcloud to estimate the odometry. I would like to extend this guess by using GPS (as in kitti available). From this i hope to get at map, that is georeferenced so that i can compare it to other localization-algorithm online. Or is there another way to georeference the surfel-map which is delivered as result of the SLAM?

Another question: can there also be an residual added to the optimzation to use landmarks? 

Thanks again for the great work! Looking forward to oyu answers!
Best regards,
Sven

Chen-Xieyuanli, 2020-04-27:
Hey @Soini3913,

Thank you for using our code.

For the map saving question, unfortunately, we don't have any plan to implement that function currently. However, there are some ways you could try to get or reconstruct the map by using our code, and more details could be found [here](https://github.com/jbehley/SuMa/issues/13).

For adding additional constraints to the optimization, there are also several ways to go. For example, you could add more constraints in the ICP process like what we've done in the semantic ICP. You could also add extra constraint nodes in the graph-optimization. Currently, we only optimize the pose graph once we confirm a loop closure, and you could also exploit better strategies there.

Hope this helps!

Soini3913, 2020-04-27:
Hi,

yes, this helps alot. Thanks for the input. :)

Chen-Xieyuanli, 2020-04-27:
Hey @Soini3913, 

I'm very happy it helps. Since we currently cannot provide any further information about this issue, I would like to close it. Please feel free to ask me to reopen it if needed.

I will also appreciate it a lot if you star&fork this repo or cite the related paper if possible.

Best,

closed 2020-04-27 by Chen-Xieyuanli