PRBonn/semantic_suma #23: The build step

Juuustin, 2020-04-29:
Recently I am trying to use SUMA to have a test, it's a little bit hard for me to build it.

Is there any video or detailed tutorial about the build step?

Chen-Xieyuanli, 2020-04-29:
Hey @Juuustin, we don't really have a video tutorial for building SuMa, but all needed steps have been explained well in the readme of [SuMa](https://github.com/jbehley/SuMa). 

For SuMa++, you will need more steps for building the [rangnet_lib](https://github.com/PRBonn/rangenet_lib), which you could find more details there.

Juuustin, 2020-05-01:
Hi， thank you for your reply,.
Is it possible to save the point cloud map showed in Rviz into .txt file?

Chen-Xieyuanli, 2020-05-02:
Hey @Juuustin, unfortunately, we don't have any plan to implement that function currently. However, there are some discussions about that and ways you could try to get or reconstruct the map by using our code. For more information, please find in previous discussions #21.

Juuustin, 2020-05-05:
Thank you for your reply

closed 2020-05-05 by Juuustin