PRBonn/semantic_suma #24: How to install opengl >= 4.0?

tym2103, 2020-06-20:
Ubuntu 16.04 default opengl version is 3.3.
How to update?
and when I finish install NVIDIA driver.
I can't get opengl info
**root@ubuntu:~# glxinfo | grep "version"
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL version string: 1.4 (2.1 Mesa 10.5.4)**

Chen-Xieyuanli, 2020-06-20:
Hey @tym2103, to install opengl in Ubuntu you could find the information on their official website [here](https://www.khronos.org/opengl/wiki/Getting_Started).

tym2103, 2020-06-21:
thank for @Chen-Xieyuanli you reply.
I follow the step,but the official way also let me download same NVIDA driver.
I think I can only read the code without debug.
and by the way, can you tell me where the follow formula use in the code

[image]

Chen-Xieyuanli, 2020-06-22:
> thank for @Chen-Xieyuanli you reply.
> I follow the step,but the official way also let me download same NVIDA driver.
> I think I can only read the code without debug.
> and by the way, can you tell me where the follow formula use in the code
> 
> [image]

Hey @tym2103, you could find the semantic icp part in the shader file Frame2Model_jacobians.geom

tym2103, 2020-06-29:
thank for @Chen-Xieyuanli  you reply.
does SM represent t-1 timestamp semantic map？
and what dose P(u|yu) mean?

Chen-Xieyuanli, 2020-06-30:
> thank for @Chen-Xieyuanli you reply.
> does SM represent t-1 timestamp semantic map？
> and what dose P(u|yu) mean?

SM is a rendered semantic representation of the Model which includes information not only from the last frame but also from history frames. 

We didn't use P(u|yu) in this work. I guess you may ask about P(yu|u), which means the probability of the semantic class giving a vertex.

I hope this helps!

tym2103, 2020-06-30:
@Chen-Xieyuanli 感谢大佬的耐心回复，下面的问题我不太会用英文描述，看到你的CSDN博客知道你懂中文，我就直接用中文说了，^_^。
SD的观察视角是当前的局部坐标，SM的观察视角是世界坐标（也就是起点的坐标）还是t-1或者t时刻的局部坐标？
然后下面P(yu|u)我说一下我的理解，假设语义地图u的大小为100*200，然后车这一类占了10个点，那么P(yu|u)的大小是不是就为10/(100*200),这样理解对吗？

Chen-Xieyuanli, 2020-06-30:
> @Chen-Xieyuanli 感谢大佬的耐心回复，下面的问题我不太会用英文描述，看到你的CSDN博客知道你懂中文，我就直接用中文说了，^_^。
> SD的观察视角是当前的局部坐标，SM的观察视角是世界坐标（也就是起点的坐标）还是t-1或者t时刻的局部坐标？
> 然后下面P(yu|u)我说一下我的理解，假设语义地图u的大小为100_200，然后车这一类占了10个点，那么P(yu|u)的大小是不是就为10/(100_200),这样理解对吗？

No problem! We can have a further discussion via e-mail in Chinese. Here is my email address: xieyuanli.chen@igg.uni-bonn.de

Since the discussion is more about the details of SuMa++ paper, I am going to close this issue and will reply to you via email.

Thank you for following our work and look forward to your email!

closed 2020-06-30 by Chen-Xieyuanli