PRBonn/semantic_suma #25: when i execute,Segfault happend

ACFFF, 2020-07-21:
Hi chen,
when i try to use suma++,i got some trouble.can you help me to solve it?
my gpu is rtx 2070 SUPER,my computer environment is
Driver Version: 440.100      CUDA Version: 10.1  cudnn:7.5.0 TensorRT-5.1.2.2
I configure my environment according to the readme.Can be compiled normally.
my system version is ubuntu18.04.the dependencies libqt5libqgtk2 is replaced by qt5-style-plugins.
`OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce RTX 2070 SUPER/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
段错误 (核心已转储)`
sometimes is
`OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce RTX 2070 SUPER/PCIe/SSE2
段错误 (核心已转储)`
I use the gdb tool to locate the segmentation fault as follows:
`GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file visualizer 
Reading symbols from visualizer...done.
(gdb) run
Starting program: /home/darren/ros_projects/rangenet_ws/src/semantic_suma/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffbfe89700 (LWP 14197)]
[New Thread 0x7fffb59a7700 (LWP 14198)]
[New Thread 0x7fffb51a6700 (LWP 14199)]
[New Thread 0x7fffaef60700 (LWP 14200)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce RTX 2070 SUPER/PCIe/SSE2

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
0x00005555555b3a25 in _mm256_store_ps (__A=..., __P=<optimized out>) at /usr/lib/gcc/x86_64-linux-gnu/7/include/avxintrin.h:880
880	  *(__m256 *)__P = __A;
(gdb) bt
#0  0x00005555555b3a25 in _mm256_store_ps (__A=..., __P=<optimized out>) at /usr/lib/gcc/x86_64-linux-gnu/7/include/avxintrin.h:880
#1  0x00005555555b3a25 in Eigen::internal::pstore<float, float __vector(8)>(float*, float __vector(8) const&) (from=..., to=<optimized out>)
    at /usr/local/include/eigen3/Eigen/src/Core/arch/AVX/PacketMath.h:251
#2  0x00005555555b3a25 in Eigen::internal::pstoret<float, float __vector(8), 32>(float*, float __vector(8) const&) (from=..., to=<optimized out>)
    at /usr/local/include/eigen3/Eigen/src/Core/GenericPacketMath.h:474
#3  0x00005555555b3a25 in Eigen::internal::assign_op<float, float>::assignPacket<32, float __vector(8)>(float*, float __vector(8) const&) const (this=<optimized out>, b=..., a=<optimized out>) at /usr/local/include/eigen3/Eigen/src/Core/functors/AssignmentFunctors.h:28
#4  0x00005555555b3a25 in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<float, float>, 0>::assignPacket<32, 32, float __vector(8)>(long, long) (this=<synthetic pointer>, this=<synthetic pointer>, col=<optimized out>, row=<optimized out>) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:652
#5  0x00005555555b3a25 in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<float, float>, 0>::assignPacketByOuterInner<32, 32, float __vector(8)>(long, long) (inner=0, outer=0, this=<synthetic pointer>) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:666
#6  0x00005555555b3a25 in Eigen::internal::copy_using_evaluator_innervec_CompleteUnrolling<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<float, float>, 0>, 0, 16>::run(Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<float, float>, 0>&) (kernel=<synthetic pointer>...) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:274
#7  0x00005555555b3a25 in Eigen::internal::dense_assignment_loop<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<float, float>, 0>, 3, 2>::run(Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<float, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<float, float>, 0>&) (kernel=<synthetic pointer>...) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:434
#8  0x00005555555b3a25 in Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<float, float> >(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<float, float> const&) (func=..., src=..., dst=...) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:741
#9  0x00005555555b3a25 in Eigen::internal::Assignment<Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<float, float>, Eigen::internal::Dense2Dense, void>::run(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<float, float> const&) (func=..., src=..., dst=...) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:879
#10 0x00005555555b3a25 in Eigen::internal::call_assignment_no_alias<Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<float, float> >(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<float, float> const&) (func=..., src=..., dst=...) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:836
#11 0x00005555555b3a25 in Eigen::internal::call_assignment<Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<float, float> >(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<float, float> const&, Eigen::internal::enable_if<!Eigen::internal::evaluator_assume_aliasing<Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::internal::evaluator_traits<Eigen::Matrix<float, 4, 4, 0, 4, 4> >::Shape>::value, void*>::type) (func=..., src=..., dst=...) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:804
#12 0x00005555555b3a25 in Eigen::internal::call_assignment<Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::Matrix<float, 4, 4, 0, 4, 4> >(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&) (src=..., dst=...) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:782
#13 0x00005555555b3a25 in Eigen::PlainObjectBase<Eigen::Matrix<float, 4, 4, 0, 4, 4> >::_set<Eigen::Matrix<float, 4, 4, 0, 4, 4> >(Eigen::DenseBase<Eigen::Matrix<float, 4, 4, 0, 4, 4> > const&) (other=..., this=0x555555f883b0) at /usr/local/include/eigen3/Eigen/src/Core/PlainObjectBase.h:714
---Type <return> to continue, or q <return> to quit---
#14 0x00005555555b3a25 in Eigen::Matrix<float, 4, 4, 0, 4, 4>::operator=(Eigen::Matrix<float, 4, 4, 0, 4, 4> const&) (other=..., this=0x555555f883b0)
    at /usr/local/include/eigen3/Eigen/src/Core/Matrix.h:208
#15 0x00005555555b3a25 in ViewportWidget::ViewportWidget(QWidget*, QGLWidget const*, QFlags<Qt::WindowType>) (this=0x555555f877d0, parent=<optimized out>, shareWidget=<optimized out>, f=...) at /home/darren/ros_proj [... cut ...]
    at /home/darren/ros_projects/rangenet_ws/build/semantic_suma/ui_visualizer.h:939
#17 0x00005555555c7811 in VisualizerWindow::VisualizerWindow(int, char**) (this=0x7fffffffd280, argc=<optimized out>, argv=<optimized out>)
    at /home/darren/ros_projects/rangenet_ws/src/semantic_suma/src/visualizer/VisualizerWindow.cpp:35
#18 0x000055555557cb02 in main(int, char**) (argc=<optimized out>, argv=0x7fffffffdbf8)
    at /home/darren/ros_projects/rangenet_ws/src/semantic_suma/src/visualizer/visualizer.cpp:19
`
What should I do? Is it an environmental configuration problem?

jbehley, 2020-07-21:
Please ensure that you have compiled gtsam with the option `USE_SYSTEM_EIGEN`, since it might give a clash between your and the version of gtsam that causes the problem:

```bash
mkdir build && cd build
cmake -DGTSAM_USE_SYSTEM_EIGEN=ON ..
sudo make install
```

Let us know if you still have trouble.

ACFFF, 2020-07-21:
> Thanks for your very detailed report on the problem.
> 
> Please ensure that you have compiled gtsam with the option `USE_SYSTEM_EIGEN`, since it might give a clash between your and the version of gtsam that (probably) causes the problem:
> 
> ```shell
> mkdir build && cd build
> cmake -DGTSAM_USE_SYSTEM_EIGEN=ON ..
> sudo make install
> ```
> 
> Let us know if you still have trouble.

Thank you for your prompt answer.i try your way right now.but it does't work.
but the gdb result has changed.
`
(gdb) bt
#0  0x00007ffff7b74d67 in _mm256_load_pd (__P=0x5555561e6310) at /usr/lib/gcc/x86_64-linux-gnu/7/include/avxintrin.h:862
#1  0x00007ffff7b74d67 in Eigen::internal::pload<double __vector(4)>(Eigen::internal::unpacket_traits<double __vector(4)>::type const*) (from=0x5555561e6310)
    at /usr/local/include/eigen3/Eigen/src/Core/arch/AVX/PacketMath.h:215
#2  0x00007ffff7b74d67 in Eigen::internal::ploadt<double __vector(4), 32>(Eigen::internal::unpacket_traits<double __vector(4)>::type const*) (from=0x5555561e6310)
    at /usr/local/include/eigen3/Eigen/src/Core/GenericPacketMath.h:463
#3  0x00007ffff7b74d67 in Eigen::internal::evaluator<Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >::packet<32, double __vector(4)>(long, long) const (this=0x7fffffffce50, col=<optimized out>, row=<optimized out>) at /usr/local/include/eigen3/Eigen/src/Core/CoreEvaluators.h:197
#4  0x00007ffff7b74d67 in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>::assignPacket<32, 32, double __vector(4)>(long, long) (this=this@entry=0x7fffffffce80, this=this@entry=0x7fffffffce80, col=<optimized out>, row=<optimized out>) at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:652
#5  0x00007ffff7b74d67 in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>::assignPacketByOuterInner<32, 32, double __vector(4)>(long, long) (inner=0, outer=0, this=this@entry=0x7fffffffce80)
    at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:666
#6  0x00007ffff7b74d67 in Eigen::internal::copy_using_evaluator_innervec_CompleteUnrolling<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>, 0, 16>::run(Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>&) (kernel=...)
    at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:274
#7  0x00007ffff7b6e275 in Eigen::internal::dense_assignment_loop<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>, 2, 2>::run(Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>&) (kernel=...)
    at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:468
#8  0x00007ffff7b6e275 in Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, 4, 4, 0, 4, 4>&, Eigen::Matrix<double, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<double, double> const&) (func=..., src=..., dst=...)
    at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:741
#9  0x00007ffff7b6e275 in Eigen::internal::Assignment<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<double, double>, Eigen::internal::Dense2Dense, void>::run(Eigen::Matrix<double, 4, 4, 0, 4, 4>&, Eigen::Matrix<double, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<double, double> const&) (func=..., src=..., dst=...)
    at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:879
#10 0x00007ffff7b6e275 in Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, 4, 4, 0, 4, 4>&, Eigen::Matrix<double, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<double, double> const&) (func=..., src=..., dst=...)
    at /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:836
#11 0x00007ffff7b6e275 in Eigen::internal::call_assignment<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, 4, 4, 0, 4, 4>&, Eigen::Matrix<double, 4, 4, 0, 4, 4> const&, Eigen::internal::assign_op<double, double> const&, Eigen::internal::enable_if<!Eigen::internal::evaluator_assume_aliasing<Eigen::Matrix<double, 4, 4, 0, 4, 4>, Eigen::internal::evaluator_traits<Eigen::Matrix<double, 4, 4, 0, 4, 4> >::Shape>::value, void*>::type) (func=..., src=..., dst=...)
[... cut ...]
    at /usr/local/include/eigen3/Eigen/src/Core/Matrix.h:208
#15 0x00007ffff7b6e275 in SurfelMapping::SurfelMapping(rv::ParameterList const&) (this=0x5555561e5950, params=...) at /home/darren/ros_projects/rangenet_ws/src/semantic_suma/src/core/SurfelMapping.cpp:46
#16 0x000055555557cb92 in main(int, char**) (argc=<optimized out>, argv=0x7fffffffd9d8) at /home/darren/ros_projects/rangenet_ws/src/semantic_suma/src/visualizer/visualizer.cpp:29
`
It seems to be an eigen problem, do you know how to solve it?thank you

jbehley, 2020-07-21:
We saw problems with different Eigen Versions, see https://github.com/PRBonn/semantic_suma/issues/14#issuecomment-598629109 and https://github.com/jbehley/SuMa/issues/20#issuecomment-563201525.

Sometimes going back do a different version of gtsam worked. Like https://github.com/jbehley/SuMa/issues/17#issuecomment-528217739 reported this.

Let me know if the switch of the version of gtsam or Eigen did help. Sorry for the trouble.

ACFFF, 2020-07-22:
> We saw problems with different Eigen Versions, see [#14 (comment)](https://github.com/PRBonn/semantic_suma/issues/14#issuecomment-598629109) and [jbehley/SuMa#20 (comment)](https://github.com/jbehley/SuMa/issues/20#issuecomment-563201525).
> 
> Sometimes going back do a different version of gtsam worked. Like [jbehley/SuMa#17 (comment)](https://github.com/jbehley/SuMa/issues/17#issuecomment-528217739) reported this.
> 
> Let me know if the switch of the version of gtsam or Eigen did help. Sorry for the trouble.

YES,I chang the gtsam version by 4.0.0-alpha2,now it worked,thank you very much.
I suggest you can write this little hint to the readme，it helped a lot!
thank you again.

jbehley, 2020-07-22:
Very good. Thanks for your suggestion, @Chen-Xieyuanli might add a note the readme.

Since this is now resolved, I close this issue.

closed 2020-07-22 by jbehley

Chen-Xieyuanli, 2020-07-22:
@ACFFF @jbehley Thank you guys for all the information. I'll add this note to the readme soon!