PRBonn/semantic_suma #26: the effect of semantic segmentation

ACFFF, 2020-07-27:
Hi，when i run semantic_suma on my computer，the effect of semantic segmentation is not the same as the your demo，This is the effect I run:
[image]
compare with yours:
[image]

i used the darknet53.

jbehley, 2020-07-27:
Please see the solutions provided in https://github.com/PRBonn/rangenet_lib/issues/9 related to Nvidia's RTX Series which are as follows:

1.The rightcombination of CUDA and TensorRT.
2. Others had success with using another ONNX version.
3. Finally, a change in the codecould resolve it: https://github.com/PRBonn/rangenet_lib/issues/9#issuecomment-644628222

Hope one of the described solutions works.

ACFFF, 2020-07-27:
> Please see the solutions provided in [PRBonn/rangenet_lib#9](https://github.com/PRBonn/rangenet_lib/issues/9) related to Nvidia's RTX Series which are as follows:
> 
> 1.The rightcombination of CUDA and TensorRT.
> 2. Others had success with using another ONNX version.
> 3. Finally, a change in the codecould resolve it: [PRBonn/rangenet_lib#9 (comment)](https://github.com/PRBonn/rangenet_lib/issues/9#issuecomment-644628222)
> 
> Hope one of the described solutions works.

thank you.i used the third way to slove it.the model darknet53 can be useful in kitti velodyne while it doesnt work in my data.i will learn [https://github.com/PRBonn/lidar-bonnetal](url) to train a model that fits my own data.

Chen-Xieyuanli, 2020-07-30:
Hey @ACFFF, since the problem was solved, I'm going to close this issue.

If there is any further problem, please re-open it again.

closed 2020-07-30 by Chen-Xieyuanli