PRBonn/semantic_suma #27: The question about IMLS experiment result ?

Gatsby23, 2020-07-30:
I have noticed that you have compared your result with IMLS-SLAM, I'm really interested in that part how to reimplement the implicit surface function ? Can you help me ?

Chen-Xieyuanli, 2020-07-30:
Hey @Gatsby23, we directly compared to the results reported in the original IMLS-SLAM paper. If you want to reproduce the results you may contact the original author, [Jean-Emmanuel Deschaud](jean-emmanuel.deschaud@mines-paristech.fr).

Gatsby23, 2020-07-31:
OK, Thank you very much!

Chen-Xieyuanli, 2020-07-31:
Hey @Gatsby23, since we cannot help you any further about this issue, I'm going to close it.

If you have any further questions, please feel free to reopen it or send us emails.

closed 2020-07-31 by Chen-Xieyuanli

Gatsby23, 2020-08-01:
OK, Thank you a lot !