PRBonn/semantic_suma #28: catkin build error related to rangenet_lib

carto-zed, 2020-07-31:
Dear authors,

  I have encountered a problem while use "catkin build ......." command to build SUMA++, the error information is showed below.

  I tried OpenCV 3.3.1, 3.4.1 and 4.1.0, but the same error occurred.

  Could you please give me a hint about how to fix this error?

[semantic_suma:make] /home/zxkj/PRBonn/devel/.private/rangenet_lib/lib/librangenet_lib.so: undefined reference to `cv::String::deallocate()'
[semantic_suma:make] collect2: error: ld returned 1 exit status                 
[semantic_suma:make] CMakeFiles/visualizer.dir/build.make:531: recipe for target '/home/zxkj/PRBonn/src/semantic_suma/bin/visualizer' failed
[semantic_suma:make] make[2]: *** [/home/zxkj/PRBonn/src/semantic_suma/bin/visualizer] Error 1
[semantic_suma:make] CMakeFiles/Makefile2:299: recipe for target 'CMakeFiles/visualizer.dir/all' failed
[semantic_suma:make] make[1]: *** [CMakeFiles/visualizer.dir/all] Error 2       
[semantic_suma:make] Makefile:138: recipe for target 'all' failed               
[semantic_suma:make] make: *** [all] Error 2                                    
Failed    <<< semantic_suma                   [ 1 minute and 3.0 seconds ]     
[build] Summary: 4 of 5 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  3 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 1 minute and 24.2 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

carto-zed, 2020-07-31:
I have fix this error.
There are different versions of OpenCV in my Ubuntu OS, cmake sometimes cannot find the same version while build rangenet_lib and semantic_suma.

Solution: specify the version of OpenCV package in rangenet_lib and semantic_suma.
step 1. in rangenet_lib/CMakeLists.txt, specify the version of OpenCV in line 156, such as `find_package(OpenCV 3.3.1 COMPONENTS core viz)`
step 2. in semantic_suma/CMakeLists.txt, specify the version of OpenCV in line 11, such as `find_package(OpenCV 3.3.1 REQUIRED)`
step 3. in your workspace, `catkin clean`
step 4. in your workspace, `catkin build --save-config -i --cmake-args -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=430 -DENABLE_NVIDIA_EXT=YES `

closed 2020-07-31 by carto-zed

Chen-Xieyuanli, 2020-08-03:
Hey @carto-zed, thank you a lot for the feedback!