PRBonn/semantic_suma #29: How to run KITTI odometry dataset

guhao1997, 2020-08-05:
I downloaded KITTI odometry dataset, there are many .bin in the velodyne file, i could run one bin at one time.
How can i choose many bin at one time to make them look like a video, not images.

Chen-Xieyuanli, 2020-08-05:
Hey @guhao1997, 

Please kindly follow the example (how to run and also the gif) we provide. After loading the bin data, you just need to press the "play button" in the visualizer GUI. The semantic SLAM will then start.

Have fun!

guhao1997, 2020-08-11:
I'm so sorry for not replying so long. My issue is same with #9 , and now i have solved this problem. thanks !

Chen-Xieyuanli, 2020-08-11:
@guhao1997 I'm glad to know that it now works for you. Since the problem was solved. I'm going to close this issue.

If the problem persists, please feel free to reopen it.

closed 2020-08-11 by Chen-Xieyuanli