PRBonn/semantic_suma #30: the visualizer has no PointCloud!

shaoquanfeng, 2020-09-10:
when I run "  ./visualizer " in the terminal,  and then, I selected a ".bin" file,  like this:
``` 
$ ./visualizer 
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1050/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
Setting verbosity to: false
Trying to open model
Trying to deserialize previously stored: /media/raymond/17354422509/suma++_ws/src/semantic_suma/config/darknet53/model.trt
Successfully found TensorRT engine file /media/raymond/17354422509/suma++_ws/src/semantic_suma/config/darknet53/model.trt
Successfully created inference runtime
No DLA selected.
Successfully allocated 426755792 for model.
Successfully read 426755792 to modelmem.
Created engine!
Successfully deserialized Engine from trt file
Binding: 0, type: 0
[Dim 5][Dim 64][Dim 2048]
Binding: 1, type: 0
[Dim 20][Dim 64][Dim 2048]
Successfully create binding buffer
calibration filename: /media/raymond/17354422509/suma++_ws/src/semantic_suma/bin/calib.txt...loaded.
ground truth filename: /media/raymond/17354422509/suma++_ws/src/poses/bin.txt
1101 poses read.
Performing frame-to-model matching. 
```
the visualizer has no PointCloud !  how can i fix this?

jbehley, 2020-09-10:
Do you use your own data or KITTI data?

shaoquanfeng, 2020-09-10:
thanks your reply! yes! I used the KITTI data.

jbehley, 2020-09-10:
Good, then it should work. ;P

When you then start the processing with the "play" button nothing happens?

shaoquanfeng, 2020-09-10:
the "play" button cannot press, which is "grey".

jbehley, 2020-09-10:
Okay, then the data is not correctly loaded. It seems like that you copied the data into the bin folder? ( /media/raymond/17354422509/suma++_ws/src/semantic_suma/bin/calib.txt)

Could you please try using the "usual KITTI" folder structure, since our data loader is specifically "hard-coded" for this structure?

```
kitti
  dataset
     sequences
          01
             velodyne
                000000.bin
                000001.bin
                   ...
             calib.txt
      poses
          01.txt
```

The loader expects certain files in certain locations. the "kitti" folder can be arbitrary. I use `~\data\kitti-odometry\`, but the structure below the KITTI folder should the correct structure. (As extracted from the zip files of the Odometry Benchmark.)

shaoquanfeng, 2020-09-10:
thanks very much, this bug if fixed, but when i corrected folder structure ,  throwing another bug, like this:
 ```
$ ./visualizer 
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1050/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
Setting verbosity to: false
Trying to open model
Trying to deserialize previously stored: /media/raymond/17354422509/suma++_ws/src/semantic_suma/config/darknet53/model.trt
Successfully found TensorRT engine file /media/raymond/17354422509/suma++_ws/src/semantic_suma/config/darknet53/model.trt
Successfully created inference runtime
No DLA selected.
Successfully allocated 426755792 for model.
Successfully read 426755792 to modelmem.
Could not deserialize TensorRT engine. 
Generating from sratch... This may take a while...
Trying to generate trt engine from : /media/raymond/17354422509/suma++_ws/src/semantic_suma/config/darknet53/model.onnx
Platform DOESN'T HAVE fp16 support.
No DLA selected.
----------------------------------------------------------------
Input filename:   /media/raymond/17354422509/suma++_ws/src/semantic_suma/config/darknet53/model.onnx
ONNX IR version:  0.0.4
Opset version:    9
Producer name:    pytorch
Producer version: 1.1
Domain:           
Model version:    0
Doc string:       
----------------------------------------------------------------
WARNING: ONNX model has a newer ir_version (0.0.4) than this parser was built against (0.0.3).
 ----- Parsing of ONNX model /media/raymond/17354422509/suma++_ws/src/semantic_suma/config/darknet53/model.onnx is Done ---- 
Success picking up ONNX model
Failure creating engine from ONNX model
Current trial size is 8589934592
Failure creating engine from ONNX model
Current trial size is 4294967296
Failure creating engine from ONNX model
Current trial size is 2147483648
Failure creating engine from ONNX model
Current trial size is 1073741824
Failure creating engine from ONNX model
Current trial size is 536870912
Failure creating engine from ONNX model
Current trial size is 268435456
Failure creating engine from ONNX model
Current trial size is 134217728
Failure creating engine from ONNX model
Current trial size is 67108864
Failure creating engine from ONNX model
Current trial size is 33554432
Failure creating engine from ONNX model
Current trial size is 16777216
Failure creating engine from ONNX model
Current trial size is 8388608
Failure creating engine from ONNX model
Current trial size is 4194304
Failure creating engine from ONNX model
Current trial size is 2097152
Failure creating engine from ONNX model
Current trial size is 1048576
terminate called after throwing an instance of 'std::runtime_error'
  what():  ERROR: could not create engine from ONNX.
Aborted (core dumped)
```

Chen-Xieyuanli, 2020-09-10:
Hey @shaoquanfeng, since you opened another issue at #31 for the second problem, I would like to close this one.

closed 2020-09-10 by Chen-Xieyuanli

wwt669, 2022-11-18:
Hi，can you help me? When i open the velodyne bin in the visualizer,it happen this issue :
./visualizer
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3090/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
Gtk-Message: 10:38:18.992: GtkDialog mapped without a transient parent. This is discouraged.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Can't open cfg.yaml from /path/to/rangenet/model/arch_cfg.yaml
已放弃 (核心已转储)
Thank you for your help

boyang9602, 2024-08-24:
@wwt669 You need to config the path of the model, which is at the last line of config/default.xml, as shown below. 
https://github.com/PRBonn/semantic_suma/blob/531954dd602ef1803707a29b9f4a5eb50cdb07de/config/default.xml#L85

It should be the darknet's path. 

For anyone using docker, just set it as `/catkin_ws/src/semantic_suma/darknet53/`. Note the code is built in the docker image, and the code inside the docker is cloned from GitHub directly, not the locally cloned code. So you need to modify it inside the docker container, not the cloned code.