PRBonn/semantic_suma #33: Hey @mazhenq, please find the kitti odometry dataset here: http://www.cvlibs.net/datasets/kitti/eval_odometry.php

mazhenq, 2020-10-20:
Hey @mazhenq, please find the kitti odometry dataset here: http://www.cvlibs.net/datasets/kitti/eval_odometry.php

_Originally posted by @Chen-Xieyuanli in https://github.com/PRBonn/semantic_suma/issues/32#issuecomment-712199186_

mazhenq, 2020-10-20:
But the Chinese website can not be downloaded, can you give me a Baidu network disk link about the KITT dates, please , thank you very much

Chen-Xieyuanli, 2020-10-20:
@mazhenq, I don't think we have the copyright to share the KITTI dataset directly by us, but you could definitely download it in China in different ways.

Here are some examples:
https://zhuanlan.zhihu.com/p/78451008
https://blog.csdn.net/weixin_43152152/article/details/99833631

Since downloading the KITTI dataset is out of the scope of our method, I could not provide any further help on that.
I'll therefore close this issue.

closed 2020-10-20 by Chen-Xieyuanli