PRBonn/semantic_suma #34: OPEN_GL and NVIDIA

jhjun724, 2020-10-22:
Hi! I'm really interested in your SUMA and SUMA++ and I'm trying to run it on my computer.
It was successful to build. Thank you for your posts.
But there was a problem in visualizer.
I think this problem caused with my computer(without graphic card).
I want to ask you that my guess is right.
```
$ glxinfo | grep "version"
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.6
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 4.60
OpenGL version string: 3.0 Mesa 20.0.8
OpenGL shading language version string: 1.30
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.0.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix,
```
Here is the result of version check and
`
~/catkin_ws/src/SuMa/bin$ ./visualizer 
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 530 (SKL GT2)
Segmentation fault (core dumped)
`
this is the error.

Thank you.

Chen-Xieyuanli, 2020-10-22:
Hey @jhjun724, I never tried our method without an Nvidia graphic card (facepalm). 
Since our method is based on OpenGL shaders, I guess it would be very hard to make it work without a graphic card.

@jbehley Jens, do you have any comments on this?

jhjun724, 2020-10-26:
Thank you for your answer!

closed 2020-10-26 by jhjun724

lifangyan16, 2022-02-07:
When you install Nvidia driver, don't add '-no-opengl-files'. When you reboot, you will find you can't open you ubuntu. This is true. 
[image]
Here, don't select 'Ubuntu' but 'Ubuntu on Wayland'
Then, set 'xorg.conf' like this:
https://forums.developer.nvidia.com/t/opengl-doesnt-use-nvidia-drivers-on-ubuntu-18-04-uses-llvmpipe/126084/4
Reboot and you can select 'Ubuntu' and can go in!
Next, you need 
sudo apt install nvidia-settings
sudo apt install nvidia-prime
When you input 'nvidia-settings' you can find this:
[image]
[image]
[image]
Congratulations!