PRBonn/semantic_suma #35: Does loop use overlap?

hitbg-yjm, 2020-12-22:
Thanks for your great work. In PRBonn/OverlapNet , it point out that overlapnet used in suma++, but I don't find overlap using in SurfelMapping::checkLoopClosure().  Loop according to residual.

Chen-Xieyuanli, 2020-12-22:
Hey @hitbg-yjm,

Thanks for following our work!

To keep both repos standalone, we currently have no plan to release the joint part. SuMa++ therefore still uses the distance-based method for finding loops. But you could easily try to combine them by easily modifying the SurfelMapping::checkLoopClosure() to use the results from OverlapNet instead of distances.

I hope my answer helps!

hitbg-yjm, 2020-12-22:
Thank you for your reply. It makes sense!

closed 2020-12-22 by hitbg-yjm