PRBonn/semantic_suma #38: confused with 天河

DayDayUpUpUp666, 2021-03-26:


DayDayUpUpUp666, 2021-03-26:
thanks for open-source the great job suma++！
I‘m confused the result. Why the resulting semantic map is black(fig 1) , not the semantic colors , just like it in your paper(fig 2)
[image]-fig1
[image]-fig2

DayDayUpUpUp666, 2021-03-26:
[image]
fig1

[image]
fig2

Chen-Xieyuanli, 2021-03-26:
Hey @DayDayUpUpUp666, thanks for using our code!

It seems a problem with semantic segmentation. You may first check the rangenet_lib: https://github.com/PRBonn/rangenet_lib

DayDayUpUpUp666, 2021-03-26:
Thanks for reply @Chen-Xieyuanli ！
I think the semantic segmentation is right , but when i want to look the whole semantic map (i choose the 'show history'), it's black 
[image] this figure is the segmentation result

[image]
this is the 'show history' , the semantic segmentation becomes black

Chen-Xieyuanli, 2021-03-26:
Show history will only show the raw point clouds with intensity information. The semantics are stored in the map. To visualize the map, you may better disable the history point clouds.

To visualize the whole surfel map, you may also change the size of the submap, since, for efficiency, we only load the tile of the submap.

DayDayUpUpUp666, 2021-03-26:
Thanks！ I’ll try later. Best Wishes!

closed 2021-09-09 by Chen-Xieyuanli

kingskymoon, 2024-04-09:
Hello, I set "submap-extent" to 100, but there were other issues, and after a period of time, the segmentation results were incorrect. Have you encountered similar errors? I submitted a new issue for detail. [https://github.com/PRBonn/semantic_suma/issues/67](url)