PRBonn/semantic_suma #39: Is SuMa++ able to do the global localization (i.e. re-localization)?

WilliamWoo45, 2021-04-07:
Hi @Chen-Xieyuanli,

Thanks for providing such great project.
I'm wondering is the `SuMa++`  able to do the semantic-based **global localization** (i.e. **re-localization**) ? 

Besides, is there any plan to extend the current SuMa++ with the capability to work with our **own 3D LiDAR dataset**?

Cheers!

Best Regards,
William

Chen-Xieyuanli, 2021-04-08:
Hey @WilliamWoo45,

Thanks for following our work!

We recently have no plan to extend SuMa++ for global localization, but we do have some LiDAR-based global localization methods ([overlap-localozation](https://github.com/PRBonn/overlap_localization) and [range-mcl](https://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/chen2021icra.pdf)).

You could use SuMa++ for your own 3D LiDAR dataset, as long as you could get the point-wise semantic information for your own scan. You could either re-train RangeNet++ or use your own semantic segmentation method.

WilliamWoo45, 2021-04-08:
Hey @Chen-Xieyuanli,

Thanks a lot for your prompt response! 
Definitely I'll try the `ovelap-localization` and I'll stay tuned for your code releasing of `range-mcl`.

BTW, apart from `ovelap-localization` & [SegMap](https://github.com/ethz-asl/segmap) & [ScanContext](https://github.com/irapkaist/scancontext) & [NDT_mapping_localization](https://github.com/melhousni/ndt_mapping_localization), is there any other open source **LiDAR-based global localization algorithm** you could recommend? 

Best Regards,
William

Chen-Xieyuanli, 2021-04-09:
Hey @WilliamWoo45, they are all good work, but may focus on different tasks.

You may first specify which problem you want to solve, place recognition or metric localization.
For metric localization, you may also tell the difference between pose tracking (with an initial guess) and global localization (without an initial guess).

Since the problem we are discussing is not about SuMa++ anymore, I would like to close this issue.
If you have any further questions regarding global localization, we could discuss them via email. My email address is xieyuanli.chen@igg.uni-bonn.de

I hope my answers are clear and helpful.
I'll also appreciate it if you could star or folk our repo if you like it.

Best,
Xieyuanli

closed 2021-04-09 by Chen-Xieyuanli

WilliamWoo45, 2021-04-09:
Hi @Chen-Xieyuanli,

Thanks for your comments.
Surely we can further discuss by email. I've starred this repo.

Cheers!

Best Regards,
William