PRBonn/semantic_suma #41: Can't visualize on my own dataset

xdtzzz, 2021-05-14:
Hi, Thank you for sharing the source code. But I have a problem. 
I've converted my dateaset into . Bin files, but when I run it, the visualization doesn't show anything like this. What's the matter.
[image]

xdtzzz, 2021-05-15:
I output the x, y, z, and intensity in the kitti bin file and my own bin file in the source program, and found that the formats of the two are the same. Why can’t I see anything in the visual interface? 
[image]

Kitti bin on the left, mine on the right .

Chen-Xieyuanli, 2021-05-17:
Hey @xdtzzz, thanks for following our work.

Could you please check "show normal map" and use "show map" with colors of something else instead of semantic?

xdtzzz, 2021-05-17:
Thank you for your  reply,  @Chen-Xieyuanli.
I changed the sensor information in the configuration file and it can run, but the effect is not very good. It should be a semantic problem. Next, I will train the dataset.
Thanks you!

closed 2021-05-20 by xdtzzz