PRBonn/semantic_suma #51: Formula error(Efficient Surfel-Based SLAM using 3D Laser Range Data in Urban Environments formula 10)

StephenYang190, 2021-12-21:
Hello, I am recently reading your work and I had a confusion about the paper Efficient Surfel-Based SLAM using 3D Laser Range Data in Urban Environments. Formula 10 use clamp operation to restrict the range of the surfels. As you mentioned in paper, clamp(x, u, l) = min(u, max(x, l)) and the u = 0.5 and l = 10 respectively. In any case, the result of clamp operation is 0.5.
My version of the paper is downloaded from google scholar as: http://roboticsproceedings.org/rss14/p16.pdf.

jbehley, 2021-12-21:
That's true, there is a mistake in the formula, as noted here:
https://github.com/jbehley/SuMa/issues/31#issuecomment-790445034

Sorry for the confusion.

StephenYang190, 2021-12-21:
> 

Thanks for your reply and kindly resite your previous answer. I will close the issuse.

closed 2021-12-21 by StephenYang190