PRBonn/semantic_suma #52: different result on suma++ and rangenet++

xdtzzz, 2021-12-23:
Thanks for your code!
When I run rangenet++ on my trained dataset, the result is almost correct, but when I run suma++ through the model.trt generated by rangenet++, the result has many wrong label, like the car , the top of the wall, and the people .
In my opinion, if the result of rangenet++ is correct, then suma++ should be correct, too.
What's the problem?
This is the result of rangenet++.
[image]
This is the result of suma++.
[image]

Chen-Xieyuanli, 2021-12-30:
Hey @xdtzzz, thanks for using our code.

The noisy map results may be caused by the wrong prediction and inaccurate poes estimation.

For example, the points of a car were predicted to be both car (blue) and house (yellow) in the segmentation results.
When aggregating all the predictions together, it will become more obvious.

xdtzzz, 2021-12-30:
谢谢您的回复！
您的意思是，每个类别都必须要分割的十分精确才可以使suma+达到良好的效果吗？
在rangenet运行，如果一个物体里大部分点云被成功的预测了，少量点云是错误的，就会导致suma+里面很多点云都是错误的情况吗

发自我的iPhone

------------------ Original ------------------
From: Xieyuanli Chen ***@***.***&gt;
Date: Thu,Dec 30,2021 3:54 PM
To: PRBonn/semantic_suma ***@***.***&gt;
Cc: xdtzzz ***@***.***&gt;, Mention ***@***.***&gt;
Subject: Re: [PRBonn/semantic_suma] different result on suma++ and rangenet++(Issue #52)

Chen-Xieyuanli, 2021-12-30:
Q: Do you mean that SuMa++ needs very good semantic segmentation results to work well?
A: Yes, SuMa++ itself can not improve the semantic segmentation results, but only use the semantics to improve the SLAM results. Wrong predictions will lead to a wrong semantic map. We originally use RangeNet++ which achieved an IOU of 52.2 and worked well. The segmentation results seem not as good as the original RangeNet++. What's your segmentation performance in terms of IOU? The current SOTA one is of IOU 70, and you may use the SOTA semantics to get a better semantic map.

是的，SuMa++对semantic segmentation本身是没有提升的，只是利用语义信息来提高SLAM的精度，如果语义信息不准，语义地图也是会不准的。
原文用的rangenet++的IOU是52.2是可以工作的，当前的语义分割结果似乎比原文差，请问现在语义分割的IOU精度是多少？ 目前性能最好的语义分割算法性能是70IOU，可以尝试使用更好的语义分割算法。

xdtzzz, 2022-01-10:
我现在语义分割的iou精度挺高的，因为标注的类别比较少，平均达到了0.8以上

xdtzzz, 2022-02-21:
Results on rangenet++, most of them are correct. But on suma++, there are many wrong semantic points. For example, the ground points build on the top of the wall on suma++, but on rangenet++, the wall doesn't have ground points.
rangenet++:
[image]
suma++:
[image]

xdtzzz, 2022-02-21:
where is "reopen" button.... # @Chen-Xieyuanli

closed 2024-03-12 by jbehley