PRBonn/semantic_suma #54: Pls, how do I save the global map?

Jaywxy, 2022-03-30:
I'm able to run the program properly, but I want to get the final global map. How do I modify the code? Or is there an interface to save global maps?

Chen-Xieyuanli, 2022-03-30:
Hey @Jaywxy, thanks for using our code. We so far didn't implement a function to save the global map, but you could have a look at variable surfels_ which stores all the surfels of the map.

You could get the surfels_ by: https://github.com/PRBonn/semantic_suma/blob/b58d880641e9af695caf68c3ac6bc883b2783a38/src/core/SurfelMap.h#L67

Jaywxy, 2022-03-30:
ok.Thanks for your reply.

closed 2022-04-30 by Chen-Xieyuanli

kslam1, 2022-11-07:
I want to remove all movable objects ，But I don't know where to add the code. Can you tell me？Thanks

kingskymoon, 2024-05-03:
> Hey @Jaywxy, thanks for using our code. We so far didn't implement a function to save the global map, but you could have a look at variable surfels_ which stores all the surfels of the map.
>
> You [...]

Hello, thank you for proposing a solution to save the point cloud. I want to save the color point cloud. I modified `VisualizerWindow:: renderMaps()`, but the result is not correct. All colors are similar to gray, (40, 40, 40)
```
        int r =  surfel_it->r*255.0;
        int g =  surfel_it->g*255.0;
        int b =  surfel_it->b*255.0;
        float w =  surfel_it->w;
        points_file << pose_world.x() << " " << pose_world.y() << " " << pose_world.z() <<" "<<surfel_it->color<<" "<<r<<" "<<g<< " "<<b<<" "<<w<< std::endl;
```