PRBonn/semantic_suma #59: save poses

pikaqiu0131, 2023-02-15:
hello, when I save poses ,there is an error:

terminate called after throwing an instance of 'std::runtime_error'
  what():  Unknown name for calibration matrix.
已放弃 (核心已转储)

do you know why? thanks!

Chen-Xieyuanli, 2023-02-15:
Thanks for using our code. As the error says, the calibration matrix is not correctly loaded.
You may check whether the calib.txt file is at the sequence folder.

pikaqiu0131, 2023-02-16:
Hello, I use the Save positions in the upper left corner of the GUI to save after each run, and then set my data file as follows：
--data 
    --08
          --velodyne
          --calib.txt
        --times.txt
        --image2
        --image3
   --calib.txt

after run the whole, I save poses, then the error turn out:
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unknown name for calibration matrix.
已放弃 (核心已转储)

pikaqiu0131, 2023-02-16:
for the calib.txt ,I think I have loaded?
Successfully create binding buffer
calibration filename: /home/sunyuhua/data/sequences/08/calib.txt...loaded.
ground truth filename: /home/sunyuhua/data/poses/08.txt
4071 poses read

Chen-Xieyuanli, 2023-02-16:
You may futher check the save_poses function here:
https://github.com/PRBonn/semantic_suma/blob/99df4940eb6b1d5a3fcb463b33a843514d48bfda/src/visualizer/VisualizerWindow.cpp#L848

BIT-TYJ, 2023-02-16:
I replaced the save_poses function https://github.com/PRBonn/semantic_suma/blob/99df4940eb6b1d5a3fcb463b33a843514d48bfda/src/visualizer/VisualizerWindow.cpp#L848
 in suma++ with the corresponding part https://github.com/jbehley/SuMa/blob/1beb131cbad33edf157d0a2333fb6193ffd6cc23/src/visualizer/VisualizerWindow.cpp#L840 in suma,  and found it solved the problem.

Note that the poses output by suma and suma++ are in the different coordinates, it seems that the poses output by suma are in the same coordinate system as the ground-truth of kitti, which should make it easier to compare pose errors.

Chen-Xieyuanli, 2023-02-16:
Dear @BIT-TYJ, thanks a lot for your feedback!
Since the problem is solved, I will close this issue and also label it as a good start issue for other users.

closed 2023-02-16 by Chen-Xieyuanli

BIT-TYJ, 2023-02-16:
Glad to solve a problem!