PRBonn/semantic_suma #63: 对‘glow::glTranslate(float, float, float)’未定义的引用

ZT-AIA, 2024-02-19:
Hello, I've recently encountered some problems using this awesome project.
Computer environment : cuda 10.1+cudnn 7.6.2+tensort5.1.5
Prerequisites : Installation rangenet++ is installed and running successfully.

Three errors (the first two have been resolved):
1. clone glow and semantic_suma compiles and runs, he starts with the error ;Unknown CMake command "COMPILE_SHADERS"
 I passed  these solved it  by adding the following to CMakeList.txt:;
```
include(/home/gwz/catkin_rangenet/src/glow/cmake/GenCppFile.cmake)
include(/home/gwz/catkin_rangenet/src/glow/cmake/GlowShaderCompilation.cmake); 
```
2. but then another error occurred: fatal error: glow/GlBuffer.h: no such file or directory.
I solved it by adding the following to CMakeList.txt:
```
include_directories("/home/gwz/catkin_rangenet/src/glow/")
include_directories("/home/gwz/catkin_rangenet/src/glow/include/")
include_directories("/home/gwz/catkin_rangenet/src/glow/src/")
include_directories("/home/gwz/catkin_rangenet/src/glow/src/glow/")
```
3. This last problem (below) I don't know what to do, please help!
 ```
MakeFiles/visualizer.dir/src/io/SimulationReader.cpp.o：在函数‘SimulationReader::SimulationReader(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’中：
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:72：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:73：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:73：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:74：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:75：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:75：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:76：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:76：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:77：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:77：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:78：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:78：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:79：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:79：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:81：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:81：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:82：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:82：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:83：对‘glow::glRotateZ(float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:83：对‘glow::glTranslate(float, float, float)’未定义的引用
/home/gwz/catkin_rangenet/src/semantic_suma/src/io/SimulationReader.cpp:84：对‘glow::glRotateZ(float)’未定义的引用
....
```

Zhangjyhhh, 2024-02-21:
have you solved this problem ,i met the same question

jbehley, 2024-02-21:
I will have a look in the next days to replace the catkin-based setup with a proper cmake-only compile. I changed the way that glow is build now and that seems to broke the compilation here.

ZT-AIA, 2024-02-21:
I then made some changes (the absolute paths are set so you need to change them to your own), which I'm not very good at, so some of the changes may be redundant, and then it compiled and ran successfully;

```
find_package(Eigen3 REQUIRED)
find_package(X11)

include_directories("/home/gwz/catkin_rangenet/src/glow/include")
include_directories("/usr/include/EGL")
file(GLOB GLOW_SRC_FILES 
    "/home/gwz/catkin_rangenet/src/glow/src/glow/*.cpp"
    "/home/gwz/catkin_rangenet/src/glow/src/glow/util/*.cpp")

add_executable(visualizer

  src/io/KITTIReader.cpp
  src/io/SimulationReader.cpp
  src/io/RobocarReader.cpp

  # use rangenet
  src/io/RangenetAPI.cpp
  
  src/opengl/Mesh.cpp
  src/opengl/Model.cpp
  src/opengl/ObjReader.cpp

  src/util/kitti_utils.cpp
  src/util/ScanAccumulator.cpp

  ${UI_HDRS}
  ${VIZ_SHADER_SRC}
  ${GLOW_SRC_FILES}

  src/visualizer/GraphWidget.cpp
  src/visualizer/ViewportWidget.cpp
  src/visualizer/VisualizerWindow.cpp
  src/visualizer/visualizer.cpp

  )

target_link_libraries(visualizer suma ${X11_LIBRARIES} ${catkin_LIBRARIES} ${Boost_LIBRARIES} ${OPENGL_LIBRARIES} ${GLEW_LIBRARIES}  EGL gtsam  pthread robovision Qt5::OpenGL Qt5::Widgets ${OpenCV_LIBS})
```
The complete CMakeList is as follows

```

cmake_minimum_required(VERSION 3.2.0)

project(semantic_suma)

find_package(OpenGL REQUIRED)
find_package(Qt5Gui REQUIRED) 
find_package(Qt5OpenGL REQUIRED)
find_package(GLEW REQUIRED)
find_package(Boost REQUIRED COMPONENTS filesystem system)
find_package(GTSAM REQUIRED)
find_package(OpenCV REQUIRED)
find_package(catkin COMPONENTS  rangenet_lib)
find_package(Eigen3 REQUIRED)
find_package(X11)

catkin_package(
  INCLUDE_DIRS src
  LIBRARIES suma
  CATKIN_DEPENDS
  glow
    rangenet_lib
  DEPENDS
    Boost
)

[... 158 lines cut ...]
  
  src/opengl/Mesh.cpp
  src/opengl/Model.cpp
  src/opengl/ObjReader.cpp

  src/util/kitti_utils.cpp
  src/util/ScanAccumulator.cpp

  ${UI_HDRS}
  ${VIZ_SHADER_SRC}
  ${GLOW_SRC_FILES}

  src/visualizer/GraphWidget.cpp
  src/visualizer/ViewportWidget.cpp
  src/visualizer/VisualizerWindow.cpp
  src/visualizer/visualizer.cpp

 
  )

target_link_libraries(robovision ${catkin_LIBRARIES} ${Boost_LIBRARIES} ${OPENGL_LIBRARIES} ${GLEW_LIBRARIES} ${rangenet_lib_LIBRARIES} gtsam pthread)
target_link_libraries(suma ${catkin_LIBRARIES} ${Boost_LIBRARIES} ${OPENGL_LIBRARIES} ${GLEW_LIBRARIES} ${rangenet_lib_LIBRARIES} gtsam pthread robovision)
target_link_libraries(visualizer suma ${X11_LIBRARIES} ${catkin_LIBRARIES} ${Boost_LIBRARIES} ${OPENGL_LIBRARIES} ${GLEW_LIBRARIES}  EGL gtsam  pthread robovision Qt5::OpenGL Qt5::Widgets ${OpenCV_LIBS})

```

Zhangjyhhh, 2024-02-22:
thank you ,that's really helpful

jbehley, 2024-02-25:
Short update on my efforts to make it cmake-buildable (similar to glow).

Unfortunately, the TensorRT 5+ interface changed significantly, therefore, I have to dig into the TensorRT 8.x code of `rangenet_lib` to make it possible to compile on Ubunutu 20.04 or Ubuntu 22.04. It will take a bit longer and I cannot promise something in the next few days.

Did the manual changes work for now?

Zhangjyhhh, 2024-02-26:
> Short update on my efforts to make it cmake-buildable (similar to glow).
>
> Unfortunately, the TensorRT 5+ interface changed significantly, therefore, I have to dig into the TensorRT 8.x code of [...]

it works!!!

jbehley, 2024-03-12:
As the manual changes resolves this particular problem, I will now close this issue.

I hopefully will look into getting rid of catkin with more modern tensor-rt in the future.

closed 2024-03-12 by jbehley