PRBonn/semantic_suma #64: error ld return 1 existing status

Zhangjyhhh, 2024-02-22:
help!! I use tensorrt 8.2.3+cuda11.0+cudnn8.04, i have successfully built rangent_lib,but when i built semantic suma,it happen:
```shell
/                                                                                        
[semantic_suma:make] /usr/bin/ld: warning: //home/jyzhang/anaconda3/lib/libquadmath.so.0: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010001                                                                
[semantic_suma:make] /usr/bin/ld: warning: //home/jyzhang/anaconda3/lib/libquadmath.so.0: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010002                                                                
[semantic_suma:make] /home/jyzhang/anaconda3/lib/libQt5Widgets.so.5.15.2: undefined reference to `std::pmr::monotonic_buffer_resource::~monotonic_buffer_resource()@GLIBCXX_3.4.28'                         
[semantic_suma:make] /home/jyzhang/anaconda3/lib/libQt5Core.so.5.15.2: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'                                             
[semantic_suma:make] /home/jyzhang/anaconda3/lib/libQt5Widgets.so.5.15.2: undefined reference to `vtable for std::pmr::monotonic_buffer_resource@GLIBCXX_3.4.28'                                            
[semantic_suma:make] /home/jyzhang/anaconda3/lib/libQt5Widgets.so.5.15.2: undefined reference to `std::pmr::get_default_resource()@GLIBCXX_3.4.26'                                                          
[semantic_suma:make] /home/jyzhang/anaconda3/lib/libQt5Widgets.so.5.15.2: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'                                                       
[semantic_suma:make] /home/jyzhang/anaconda3/lib/libQt5Core.so.5.15.2: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'                                              
[semantic_suma:make] collect2: error: ld returned 1 exit status                                                                                                                                             
[semantic_suma:make] CMakeFiles/visualizer.dir/build.make:804: recipe for target '/home/jyzhang/catkin_ws/src/semantic_suma/bin/visualizer' failed                                                          
[semantic_suma:make] make[2]: *** [/home/jyzhang/catkin_ws/src/semantic_suma/bin/visualizer] Error 1                                                                                                        
[semantic_suma:make] CMakeFiles/Makefile2:304: recipe for target 'CMakeFiles/visualizer.dir/all' failed                                                                                                     
[semantic_suma:make] make[1]: *** [CMakeFiles/visualizer.dir/all] Error 2                                                                                                                                   
[semantic_suma:make] Makefile:145: recipe for target 'all' failed                                                                                                                                           
[semantic_suma:make] make: *** [all] Error 2                                                                                                                                                                
Failed    <<< semantic_suma                [ 2.7 seconds ]                                                                                                                                                 
[build] Summary: 2 of 3 packages succeeded.                                                                                                                                                                
[build]   Ignored:   None.                                                                                                                                                                                 
[build]   Warnings:  3 packages succeeded with warnings.                                                                                                                                                   
[build]   Abandoned: None.                                                                                                                                                                                 
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 4.1 seconds total.   
```

Zhangjyhhh, 2024-02-22:
I solveit by deleting the build file  and rebuild it , but i do not know why

closed 2024-02-22 by Zhangjyhhh

kingskymoon, 2024-04-08:
I delete the build file，but  it doesnt work. So i delete libQt5Widgets, libQt5Core files in anaconda. (it is not a good way but work)