PRBonn/semantic_suma #68: question about GLOW

Nicoleperry, 2025-04-02:
[semantic_suma:cmake] All togheter now (libs): /usr/local/cuda-11.6/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/x86_64-linux-gnu/librt.so;nvinfer;nvinfer_plugin;nvparsers;nvonnxparser
[semantic_suma:cmake] All togheter now (inc): /usr/local/cuda-11.6/include
[semantic_suma:cmake] Enabling Nvidia OpenGL extensions.                        
[semantic_suma:cmake] CMake Error at /home/zyt/catkin_ws/src/semantic_suma/CMakeLists.txt:61 (COMPILE_SHADERS):
[semantic_suma:cmake]   Unknown CMake command "COMPILE_SHADERS".
[semantic_suma:cmake] 
[semantic_suma:cmake] 
[semantic_suma:cmake] -- Using OpenGL version 460.                              
[semantic_suma:cmake] -- Configuring incomplete, errors occurred!               
[semantic_suma:cmake] See also "/home/zyt/catkin_ws/build/semantic_suma/CMakeFiles/CMakeOutput.log".
[semantic_suma:cmake] See also "/home/zyt/catkin_ws/build/semantic_suma/CMakeFiles/CMakeError.log".
Failed    <<< semantic_suma                [ 1.0 seconds ]                     
[build] Summary: 2 of 3 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2.3 seconds total.                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.

I don't know where the problem lies, please advise me
already entered： cd ~/catkin_ws/src
git clone https://github.com/PRBonn/semantic_suma.git
git clone https://github.com/jbehley/glow.git

you1shuo1, 2025-04-12:
I met the same issue.It seems that I did not git clone the correct glow.You can try to download the glow-0.1.[https://github.com/jbehley/glow/releases/tag/v0.1](url).After this catkin build again.

you1shuo1, 2025-04-12:
By the way, it worked for me.